tinygo-org/drivers #412: irremote not working properly using Arduino Uno

LeonardsonCC, 2022-05-10:
Hey,
I'm new to embedded stuff, so maybe I have made some mistake.

Trying to make the IR Remote driver to work using an Arduino Uno, it just not received any signal.
Debugging the driver code, I notice that if I use the [Pin Mode as Input](https://github.com/tinygo-org/drivers/blob/release/irremote/receiver.go#L68), and change the [`SetInterrupt`](https://github.com/tinygo-org/drivers/blob/release/irremote/receiver.go#L77) to just call `pinChange` each 10 milliseconds, it receives the controller inputs.

There's any reason for the `SetInterrupt` and the Pin Mode as InputPullup  to not work on Arduino Uno?

Checking [Arduino docs](https://www.arduino.cc/reference/pt/language/functions/external-interrupts/attachinterrupt/), the pins 2 and 3 should be working, right?

neildavis, 2022-07-04:
Hi @LeonardsonCC which IR receiver component are you using with the Uno? Can you provide the part number and/or link to the datasheet? Thanks.

LeonardsonCC, 2022-07-04:
Hi, I believe it is this one: http://eeshop.unl.edu/pdf/VS1838-Infrared-Receiver-datasheet.pdf

neildavis, 2022-07-04:
Thanks. If I'm reading that datasheet correctly, it's an 'active low' device so the driver config should be compatible. I have a R3 Uno somewhere so will try it soon once I have done some maintenance on my tinygo setup. I'm assuming you were using tinygo `v0.23` at the time of reporting?