tinygo-org/drivers #442: Driver for GC9A01 Display

lmatte7, 2022-07-26:
Hello! I'd like to contribute a new driver for the [GC9A01](https://www.waveshare.com/1.28inch-lcd-module.htm) driver for a round display. I've started some preliminary work, but wanted to open an issue first to make sure I wasn't duplicating any other work (per the `CONTRIBUTING.md` file.

deadprogram, 2022-07-26:
Round display? Cool... Yes, please, and thank you!

rcbadiale, 2022-10-02:
Hey lmatte7, I was going to start looking into this display, did you had any progress?

If you need any help please let me know.

lmatte7, 2022-10-02:
I did make some progress! There’s a PR merged now that’s waiting for a release https://github.com/tinygo-org/drivers/pull/452

rcbadiale, 2022-10-02:
Cool! I didn't noticed it in dev, sorry for bothering!

I will take a look at what you've done.

Thank you very much for your work!

lmatte7, 2022-10-02:
No worries! I’m glad there are some other people using these cool little displays 😄

lmatte7, 2024-03-16:
Oh I didn't realize this was still open! Closing since it was resolved with #452

closed 2024-03-16 by lmatte7