tinygo-org/drivers #455: Want to change the wifi example from const to var.

sago35, 2022-09-06:
The current code is as follows

```
// access point info
const ssid = ""
const password = ""
```

SSID settings, etc., require direct modification of the code.
Changing directly increases the possibility of accidental git push.

So I suggest that it be the following
The following form can be rewritten by adding another file.

It can also be rewritten using ldflags. (`tinygo flash -target wioterminal -ldflags '-X "main.ssid=YOUR_SSID"'`)

```
// You can override the setting with the init() in another source code.
// func init() {
//    ssid = "your-ssid"
//    password = "your-password"
// }

var (
	// access point info
	ssid string
	pass string
)
```

So, I would like to rewrite examples.
What do you think?

sago35, 2022-09-06:
```
$ git grep -i "const ssid"
examples/espat/espconsole/main.go:23:const ssid = "YOURSSID"
examples/espat/esphub/main.go:21:const ssid = "YOURSSID"
examples/espat/espstation/main.go:18:const ssid = "YOURSSID"
examples/espat/mqttclient/main.go:24:const ssid = "YOURSSID"
examples/espat/mqttsub/main.go:25:const ssid = "YOURSSID"
examples/espat/tcpclient/main.go:18:const ssid = "YOURSSID"
examples/wifinina/connect/main.go:13:const ssid = ""
examples/wifinina/http-get/main.go:28:const ssid = ""
examples/wifinina/mqttclient/main.go:25:const ssid = ""
examples/wifinina/mqttsub/main.go:25:const ssid = ""
examples/wifinina/ntpclient/main.go:18:const ssid = ""
examples/wifinina/tcpclient/main.go:21:const ssid = ""
examples/wifinina/tlsclient/main.go:22:const ssid = ""
examples/wifinina/udpstation/main.go:19:const ssid = ""
examples/wifinina/webclient/main.go:19:const ssid = ""
```

deadprogram, 2022-09-07:
Very good idea!

aykevl, 2022-09-15:
Seems like a good idea!

deadprogram, 2023-02-12:
This has now been released, so closing. Thank you!

closed 2023-02-12 by deadprogram