tinygo-org/drivers #460: microbitmatrix: x and y axis inverted?

breml, 2022-09-21:
The microbitmatrix driver does not give me the expected results, if using the `SetPixel` method. The method is defined as `SetPixel(x int16, y int16, c color.RGBA)`, where I expect `x` to be the horizontal and y the vertical axis and `(0, 0)` to be in the upper left corner (if `rotation` is set to `0`, which is the default value).

This expectation comes from using the Makecode, JavaScript and Python implementations on https://makecode.microbit.org/.

From the help text of the JavaScript `led.plot` function:

```none
function led.plot(x: number, y: number): void
Turn on the specified LED using x, y coordinates (x is horizontal, y is vertical). (0,0) is upper left.

@param x — the horizontal coordinate of the LED starting at 0

@param y — the vertical coordinate of the LED starting at 0
```

or Python:

```none
def led.plot(x: int32, y: int32): None
Turn on the specified LED using x, y coordinates (x is horizontal, y is vertical). (0,0) is upper left.
```

With the following program I expect the LED in the top right corner to light (x = 4, y = 0, rotation = 0), but instead, the LED in the bottom left corner does light.

```golang
package main

import (
	"image/color"

	"tinygo.org/x/drivers/microbitmatrix"
)

var (
	display microbitmatrix.Device
	c       = color.RGBA{255, 255, 255, 255}
)

func main() {
	display := microbitmatrix.New()
	display.Configure(microbitmatrix.Config{
		Rotation: 0,
	})

	display.SetPixel(4, 0, c)
	for {
		display.Display()
	}
}
```

I have tested the above program on https://play.tinygo.org/ as well as on a microbit V1.5 and microbit V2.20 with the same results.

I quickly copied the code of the microbitmatrix driver to my project and swapped `x` and `y` in https://github.com/tinygo-org/drivers/blob/release/microbitmatrix/microbitmatrix.go on lines 31, 33 and 42, which did resolve the problem for me.

I am happy to file a PR for this.

conejoninja, 2022-09-23:
Hello, just tested it on mine (1.2? original? nor sure) and `x` `y` are swapped, not sure when/how this happened. Feel free to make a PR for this, I'll gladly review + approve it

breml, 2022-09-24:
I went back through the git history and I also was not able to figure out when this changed. Is it possible, that this bug has been in there since the beginning?

PR #462 (2022-09-24): microbitmatrix: fix inverted axis
Resolves #460
commit 98ba5a23 (2022-09-24): microbitmatrix: fix inverted axis

Resolves #460
closed 2022-10-01 by deadprogram