tinygo-org/drivers #468: wifinina: can't connect to AP 50% of the time

scottfeldman, 2022-10-05:
I'm using drivers/examples/wifinina/tcpclient/main.go example on Arduino Nano33 IoT.

I'm seeing AP connect fail 50% of the time.  In fact, it seems to toggle between pass and fail.  Usually, after tinygo flash, the AP connect will fail.  Pressing the hard reset button on the Nano33 will try the AP connect again.  This time it will connect.  Pressing the hard reset again, the AP connect will fail.  And so on...

I logged the connection status during the AP connect to see what's going on:

```
[tinygo flash]                                                                                                                                        
                                                                                                                                                    
Connecting to [my ssid]                                                                                                                                 
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS Connect Failed                         <-- failed                                                                                                      
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
STATUS Connect Failed                                                                                                                               
...                                                                                                                                                 
                                                                                                                                                    
[HARD RESET]                                                                                                                                        
                                                                                                                                                    
Connecting to [my ssid]                                                                                                                                 
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS No SSID Available                                                                                                                            
STATUS Connected                            <-- success                                                                                                        
Connected.                                                                                                                                          
192.168.1.152                                                                                                                                       
---------------                                                                                                                                     
Dialing TCP connection                                                                                                                              
...                                                                                                                                                 
                                                                                                                                                    
[HARD RESET]                                                                                                                                        
                                                                                                                                                    
Connecting to [my ssid]                                                                                                                                 
STATUS No SSID Available                                                                                                                            
[... cut ...]
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS No SSID Available
STATUS Connected                    <-- success
Connected.
192.168.1.152 
---------------
Dialing TCP connection 
...
```
I tried various delays after holding RESET low and high.  No change.

I tried RESET twice (low/high/low/high).  No change.

scottfeldman, 2022-10-28:
I tried the same wifinina/tcpclient example on Arduino Nano RP2040 and get the same results.  Connection attempts fails 50% of the time.

(wioterminal using rtl8720dn/tcpclient example connects 100% of the time, connecting the same AP as the Arduino Nano tests).

scottfeldman, 2022-11-12:
I tried using a different Wifi router, and the issue goes away.  The issue was first reported on a Starlink Wifi router, so maybe this is limited to connecting to Starlink router.

scottfeldman, 2023-01-31:
Just a note for now: this issue is fixed with the netdev branch.

deadprogram, 2023-05-14:
Also fixed by https://github.com/tinygo-org/drivers/pull/561 until that gets merged in TinyGo 0.29 release.

deadprogram, 2023-10-26:
This was already released so now closing. Thanks!

closed 2023-10-26 by deadprogram