tinygo-org/drivers #470: wifinina: http-get test fails due to too small of buffer

scottfeldman, 2022-10-05:
Running examples/wifinina/http-get example on Arduino Nano33 IoT.  Example fails with this error:
```
slice out of range : use http.SetBuf() to change the allocation to 1036 bytes or more 
```
Fix is to increase buffer size:
```
--- a/examples/wifinina/http-get/main.go
+++ b/examples/wifinina/http-get/main.go
@@ -44,7 +44,7 @@ var (
        adaptor *wifinina.Device
 )
 
-var buf [0x400]byte
+var buf [0x500]byte
```
I'll prepare a PR if the change is acceptable...

$ tinygo version
tinygo version 0.25.0 linux/arm (using go version go1.19.1 and LLVM version 14.0.0)

scottfeldman, 2023-01-31:
Just a note for now: this issue is fixed with the netdev branch.

scottfeldman, 2023-05-17:
Also fixed by #561 until that gets merged in TinyGo 0.29 release.

deadprogram, 2024-02-28:
Closing as completed in the most recent release. Thank you!

closed 2024-02-28 by deadprogram