tinygo-org/drivers #481: atlas: Atlas Scientific PH/EC sensor support

SoleimanyBen, 2022-11-11:
Was looking to add support for Atlas Scientific PH/EC sensor. 

I see no one is actively working on it, so I thought I would go ahead and take a stab at it.

PH Sensor: https://atlas-scientific.com/kits/gravity-analog-ph-kit/
EC Sensor: https://atlas-scientific.com/kits/conductivity-k-1-0-kit/

deadprogram, 2022-11-14:
Sounds great @SoleimanyBen let us know here how we can help!

SoleimanyBen, 2022-11-14:
Thanks for the reply @deadprogram! It turns out that the Atlas Scientific libraries rely on the onboard EEPROM in order to store the calibration values into it.

I already created a simple driver that interacts with the PH sensor, but it would require an external `SPI/QSPI` flash storage device in order to use it. This can be easily extended to support i2c EEPROMs, but I really think it would be better to use the onboard storage.

I was looking into possibly adding support for onboard EEPROMs. I commented on https://github.com/tinygo-org/tinygo/issues/1566#issuecomment-1312387089 and was hoping to get more insight on the comment I made about how it would be possible to implement it.

From the research I've done, it seems like it should not be too difficult to implement the EEPROM lib?