tinygo-org/drivers #483: DHT 22 not working with ESP32

donni-h, 2022-11-18:
Hey folks, 
so I was trying to hook up a DHT22 to my ESP32 Dev Module with the following basic code:

```
package main

import (
   "machine"
   "tinygo.org/x/drivers/dht"
)

func main() {
   sensor := dht.New(machine.SDA_PIN, dht.DHT22)
   _ = sensor
}
```

whatever I did calling `dht.New(machine.SDA_PIN, dht.DHT22)` caused tinygo to fail flashing with the error> `machine.UART1.Interrupt undefined (type *machine.UART has no field or method Interrupt)`. 
Is this a bug, since I've been told that ESP support in tinygo is not quite perfect yet, or is it an issue on my end?

jasdel, 2022-11-20:
I noticed this as well with the esp8266 nodeMCU board. It looks like the `dht` driver uses [machine.UART1](https://github.com/tinygo-org/drivers/blob/v0.23.0/dht/thermometer.go#L163-L164), but the esp boards only defines a [UART0](https://github.com/tinygo-org/drivers/blob/v0.23.0/dht/thermometer.go#L163-L164). Is there something special about `UART1` vs `UART0`?

Should the `dht` driver be referring to the [machine.DefaultUART](https://github.com/tinygo-org/drivers/blob/v0.23.0/dht/thermometer.go#L163-L164) package variable instead of a specific `UART#` directly?

aykevl, 2022-12-08:
That's a bug in the driver. Instead of using `UART1.Disable()` etc it should use `mask := interrupt.Disable()` and `interrupt.Restore(mask)` (from runtime/interrupt)
Feel free to send a PR to fix this issue.

commit 789bbbd8 (2022-12-20): Update DHT22 receive to use runtime/interrupt

Fixes #483 by updating the DHT22 device to use runtime/interrupt instead
of machine.UART1 for interrupt disabling while receiving signals from the
device.
closed 2023-02-12 by deadprogram