tinygo-org/drivers #489: request for ds18b20 driver

kellyfitzy, 2022-12-11:
Hi,

I'm a novice programmer looking for a tinygo ds18b20 driver to use with Raspberry Pi Pico W.   Willing to test.  Have used various ds18b20 golang drivers on Raspberry Pi Nano, etc.  Thanks in advance!

dn-kolesnikov, 2022-12-17:
I'm done my library for 1-wire
In current moment I’m adding support for ds18b20
In few days I’m going to push it to my GitHub
Tested for RP pico

dn-kolesnikov, 2022-12-26:
> Hi,
> 
> I'm a novice programmer looking for a tinygo ds18b20 driver to use with Raspberry Pi Pico W. Willing to test. Have used various ds18b20 golang drivers on Raspberry Pi Nano, etc. Thanks in advance!

https://github.com/dn-kolesnikov/tinygo-drivers.git

first commit
so far, it only works with a single sensor on the bus.
no docs, no tests. :no_good:
the rest is in the process
see example...

kellyfitzy, 2022-12-26:
Thanks, I'll check it out.

matveynator, 2023-01-05:
> https://github.com/dn-kolesnikov/tinygo-drivers.git
>
> first commit so far, it only works with a single sensor on the bus. no docs, no tests. 🙅 the rest is in the process see example...

Could you pull to tinygo-drivers when ready and stable? Thank you very much!

conejoninja, 2023-01-05:
Hello @dn-kolesnikov , thanks for your work, as @matveynator said, please make a pull request to the repo so we can properly review it.

I do have some DS18B20 laying around (the waterproof version), and will gladly review/test it.

One thing I notice, it that the ReadTemperature looks like this 
```go
func (d Device) ReadTemperature() (Temperature, error) {
```
It would be great if it has the same signature as other temperature sensor in the repository: temperature should be int32, and it should be milliCelsius instead of Celsius. That makes any temp. sensor interchangeable, and you could replace one with another without much effort.

```go
func (d *Device) ReadTemperature() (temperature int32, err error) {
```

dn-kolesnikov, 2023-01-05:
> Hello @dn-kolesnikov , thanks for your work, as @matveynator said, please make a pull request to the repo so we can properly review it.
>
> I do have some DS18B20 laying around (the waterproof [...]

ok, I'll try to do a refactoring within a couple of weeks

dn-kolesnikov, 2023-01-05:
> Could you pull to tinygo-drivers when ready and stable? Thank you very much!
Ok, I’ll do it! 
—-
до конца января постараюсь реализовать все ROM функции и зареквестить в мастер

dn-kolesnikov, 2023-01-08:
> Hello @dn-kolesnikov , thanks for your work, as @matveynator said, please make a pull request to the repo so we can properly review it.
>
> I do have some DS18B20 laying around (the waterproof [...]

Corrected the function, you can check it

conejoninja, 2023-01-08:
Hello @dn-kolesnikov ,

I went ahead and moved your code to a new branch in the drivers repository as I was having some issues with the import paths of your code : https://github.com/conejoninja/drivers/tree/onewire (this is my personal repo). I added smoke tests and added it to the README.md devices' list too. Fixed some typo in the comments/doc.

Feel free to make any modification you want to, or make your own PR. I did it because I plan to use the ds18b20 in one example at a talk at #FOSDEM (4th February), so I need this working as soon as possible.

I still can't get it to work, I'm getting the following error 

```
Read OneWire ROM
Error: OneWire. No devices on the bus.
```

I'm using Arduino Nano33 IOT, connected to pin D13 and a 4.7K resistor between D13 and 3.3V pins. My hardware setup works fine with this C library : https://github.com/milesburton/Arduino-Temperature-Control-Library but strange enough not with this other : https://github.com/matmunk/DS18B20

dn-kolesnikov, 2023-01-08:
Hi @conejoninja,
 I have uno, nano and micro boards.  I will check on these boards and let you know the results.  on Monday (09.01.2023) after 10 am (GMT+5) I will be in the office and do all the checks.

conejoninja, 2023-01-09:
After a brief talk and some more tests, the driver is working properly and what was wrong with my setup was powering the sensor directly from the board. It's still a bit strange but this is what I got:

- It works (powered by the board) with  https://github.com/milesburton/Arduino-Temperature-Control-Library (arduino library, this doesn't make sense to me)
- It does NOT work (powered by the board) with https://github.com/matmunk/DS18B20 (arduino library)
- It does NOT work (powered by the board) with this go driver
- It does NOT work with this go driver if powered by external source of 3.3V
- It WORKS with this go driver if powered by external source of 3.4V or more (up to 5V)

Thanks for your help sorting this out. Great work with this driver.

dn-kolesnikov, 2023-03-06:
Hi there!

https://github.com/dn-kolesnikov/tinygo-drivers.git

The driver functionality is ready for use. There are functions for searching devices on the bus (up to 32), reading the temperature in raw values and in Celsius /1000.
I use 5 sensors to control the electric heating of my house.
Who is ready to add this driver to the main branch?

dn-kolesnikov, 2023-03-29:
> Right now, you just create one DS18B20 device and handle the different sensors passing the romId around, could it make sense to use a DS18B20 device for each sensor? When 

I thought about how to make the implementation of various devices around the 1-Wire bus more elegant.  In the current version, we are not creating a separate DS18B20 device, we are creating an interface that matches the 1-Wire bus methods.  I wanted to combine everything into one onewire package with the implementation of a single device “Device” containing only the pin number.  since for work we need only a pin and supports, and all methods for various devices are called through a prefix, for example DS18B20_ReadTemperature () and so on ...

perhaps, with the expansion of functionality, we will do refactoring.  if there are thoughts on how to improve the existing implementation, I'm open to suggestions.

deadprogram, 2024-12-17:
Closing since we have had a `ds18b20` driver for a while now. Thanks everyone!

closed 2024-12-17 by deadprogram