tinygo-org/drivers #510: Add a driver for ams AS5600

neildavis, 2023-01-17:
Add driver support for the [ams AS5600](https://ams.com/as5600) magnetic angle measurement/rotary position sensor. It can work via I2C and/or analog/PWM. 

I'm going to be working with these in another language. I didn't see support for it in the existing drivers, or an existing issue, so opening this as a marker, to avoid duplication of effort etc. If/when I get time I may do the work if this is still unclaimed.

neildavis, 2023-02-01:
@deadprogram I have WIP for this now, so expect a PR soon

deadprogram, 2023-02-01:
Looking forward to it!

neildavis, 2023-02-07:
#520 is ready for review :)

neildavis, 2023-02-13:
@deadprogram Can I get a review please? I'm not sure who to request from, so if you've not got time yourself perhaps you could ask another kind soul? Thanks!

deadprogram, 2023-10-26:
This was already released so now closing. Thanks!

PR #520 (2023-03-21): Add support for ams AS560x on-axis magnetic rotary position sensors (#510)
Resolves #510. I think it's time to get some eyes on this :)

* Support for both AS5600 & AS5601 since they're very similar
* Only I2C interface supported. I guess Analog/PWM are too trivial to require driver support?

Issues I'd appreciate opinions on:

1. How many 'convenience' methods (wrappers around lower-level register read/write methods) should we include in the API? I've included only what I think are likely to be commonly used and/or where they can 'add value' (e.g. angle unit conversions, decoding of `STATUS`  [... cut]
closed 2023-10-26 by deadprogram