tinygo-org/drivers #540: WS2812 driver does not work with XIAO SAMD21

ysoldak, 2023-04-01:
I've tried to drive RGB Led with XIAO board (SAMD21, 48MHz chip)  and it does not work.

When trying the example, all 10 LEDs are very bright, almost white. One can see a hint of green and red, changing though.

Same example on a RP2040 (125MHz) board just works (blinking green and red).

aykevl, 2023-04-01:
I checked on the Circuit Playground Express (same chip) and it works fine.
Can you double check:
- the wires (is it really connected correctly - perhaps try another pin)
- power requirements (maybe it doesn't have enough power to drive these LEDs correctly), also note that the ground must be shared between the MCU and the LEDs.

If it's not a wire, it's hard to say what's wrong. It might be a faulty crystal oscillator or any number of other things. The way to know for sure is to connect it to a logic analyzer and look at the signal.

Something else you could try: testing the same setup but using the Arduino IDE and a library like FastLED.

ysoldak, 2023-05-17:
I've verified again, here are results:
Control (XIAO RP2040) - Correct
[image]
Experiment (XIAO SAMD21) - Too bright
[image]

Here is overview of the setup, wires indeed quite long and connections may be weak, but it works for control...
[image]

To make this work, I've added following two files to the ws2812 example.
And unmatched these tags in `others.go`

XIAO RP2040
`./examples/ws2812/xiao-rp2040.go`
```
//go:build xiao_rp2040

package main

import "machine"

var neo = machine.D8
var led = machine.LED

func init() {
	machine.NEOPIXEL_POWER.Configure(machine.PinConfig{Mode: machine.PinOutput})
	machine.NEOPIXEL_POWER.High()
}
```
`tinygo flash -target=xiao-rp2040 -size=full ./examples/ws2812/`

XIAO SAMD21
`./examples/ws2812/xiao.go`
```
//go:build xiao

package main

import "machine"

var neo = machine.D8
var led = machine.LED
```
`tinygo flash -target=xiao -size=full ./examples/ws2812/`

ysoldak, 2023-05-17:
This was just something I've observed when working on a random project that runs on RP2040 now anyway.
So this bug(?) does not block me in any way and I'm too lazy to debug this further.
Let it be more as FYI and we can look at this closer if more cases like this emerge.

P.S. LED strip we can rule out from the equation, the problem first discovered with another strip (different size and brand), same result.

aykevl, 2023-05-19:
I don't have a Xiao so unfortunately I can't debug this issue myself. But here are some other ideas to narrow down this problem:

* Can you try on another XIAO SAMD21 board (to rule out a hardware problem with the board)?
* Do you have another samd21 based board that you can try? For example, one of the "M0" boards from Adafruit.

ysoldak, 2023-05-25:
Tried with another XIAO board and another pin -- same result.
Also, tried with Arduino Nano 33 IoT board, same chip. Same bad result.
[image]

An observation: on XIAO-RP2040 all output pins work fine, except D10. In D10 case only one LED is lit up, too bright and slightly pulsing. This in contrast with SAMD21 there all leds are brightly lit up.

Code of my experiments you can find in #575 pull request.

aykevl, 2023-05-26:
I have been able to reproduce this issue using an Arduino Nano 33 IoT. I'll see what I can do.

aykevl, 2023-05-26:
Here is a fix: https://github.com/tinygo-org/tinygo/pull/3751
Can you test it on the XIAO?

aykevl, 2023-05-26:
> An observation: on XIAO-RP2040 all output pins work fine, except D10. In D10 case only one LED is lit up, too bright and slightly pulsing. This in contrast with SAMD21 there all leds are brightly lit up.

This is an unrelated issue, as it's an entirely different chip.
If this is a problem for you, you can open a new issue.

ysoldak, 2023-05-26:
>Can you test it on the XIAO?

I did, it works now, thank you.

ysoldak, 2023-05-26:
>This is an unrelated issue, as it's an entirely different chip.
If this is a problem for you, you can open a new issue.

Not a problem at all, just an observation I thought worth to share.
No fix needed as any other pin can be used for workaround.

ysoldak, 2023-05-26:
Feel free to close this as fixed when you see fit, otherwise I do that after TinyGo v0.28 is released.

ysoldak, 2023-08-02:
Fixed in TinyGo v0.28