tinygo-org/drivers #552: Add MPU6886 driver

ivoszz, 2023-04-24:
I want to add M5Stick-C to tinygo as a supported platform with its MPU6886 (also M5Stack Core/Core2) accelerometer/gyroscope as a driver. I started by adding basic functions of the driver based on the MPU6050. But I found that I don't quite understand the concept of the return values of the ReadAcceleration() and ReadRotation() functions. These functions do not work with sensitivity settings (e.g. +/- 2g, 4g, 8g, 16g) and always return scaled raw counter values. So if I set the sensitivity to 16g, the return values are not in the range +/- 1000000, but approximately +/- 125000. Is it then up to the user to scale the result correctly?

I am able to add basic functionality in a short time. But MPU6050 doesn't seem like an ideal model to me, is it possible to recommend some other accelerometer/gyroscope unit? I would like to add more advanced features like using FIFO or IRQ later. Is there a good example? I'm not very good at naming variables and functions and a template would come in handy.

I've read the terms and conditions for contributing, is there anything I should pay particular attention to? 

Thanks for any suggestions.

soypat, 2023-04-29:
Yes, the existing mpu6050 driver is extremely limited. Please refer to the rewrite here https://github.com/tinygo-org/drivers/pull/556 for a more ideal approach to implementing future IMU drivers.

This implementation does not have FIFO or IRQ but could possibly be extended.

deadprogram, 2024-12-17:
Closing since it was added already. Thanks everyone!

closed 2024-12-17 by deadprogram