tinygo-org/drivers #557: Extract real-time waits to own package

soypat, 2023-04-29:
I can't help but think maybe packages like [`i2csoft`](https://github.com/tinygo-org/drivers/blob/release/i2csoft/i2csoft_esp32.go) and others could benefit from having a package that implements real-time waits for several architectures instead of having each driver implement their own waits for every architecture.

Suggestion for API:

```go
func NewRealTimeWaiter(waitTime time.Duration) Waiter

func (w Waiter) Wait() {
    return w.wait() // architecure specific implementation.
}

type Waiter struct {
    nopcount int // Or alternatively, guard waiter data structure for every implementation
}
```

aykevl, 2023-04-29:
Sounds reasonable.
However, the current implementation is not very precise. I think it might be a good idea to implement this in assembly, like is done for the WS2812 driver to wait a very precise number of cycles without the guesstimates that are currently used (that might change as a result of compiler optimizations for example). As shown in the ws2812 driver, it is possible on most chips to wait a very specific number of cycles.

bxparks, 2023-05-01:
Maybe the following proposal goes against some design philosophy regarding TinyGo, but I would like to see a slightly different API, something that looks like this:

```go
// Return a monotonically increasing integer representing the number of microseconds
// from an arbitrary point in time. The counter wraps around to 0 upon overflow.
func micros() uint32

// Return a monotonically increasing integer representing the number of milliseconds
// from an arbitrary point in time. The counter wraps around to 0 upon overflow.
func millis() uint32
```

So I guess I'm assuming that a hardware timer is available that can interrupt the CPU and increment a counter quickly and efficiently.

We can build `delayMicros(n)` and `delayMillis(n)` on top of `micros()` and `millis()`, but not the other way around:

```go
func delayMicros(n uint32) {
  start := micros()
  while micros() - start < n {}
}

func delayMillis(n uint32) {
  start := millis()
  while millis() - start < n {}
}
```

Occasionally, my device driver needs to perform some operation while waiting for a short time interval (i.e. anywhere between 1 and 1000 microseconds) to pass. The following pseudo-code is possible with `micros()` but not with `delayMicros()`:

```go
var start uint32
for {
  elapsed := micros() - start
  if elapsed >= duration {
    // doThing1
    break
  }
  // doThing2 while waiting
}
```

I think a `uint32` instead of `Duration` (`uint64`) would save memory and CPU resources. Most TinyGo microcontrollers use 32-bit processors which handle 32-bit operations quite efficiently. Operations on `uint64` are at least 2X more expensive for additions/subtractions. For 32-bit processors without hardware multiplication and division instructions, `uint64` multiplications and divisions can be up to 10X more expensive according to my microbenchmarks. On 8-bit AVR processors, 64-bit operations are even more expensive.

In at least one of my device drivers, even a `uint32` microseconds is larger than necessary. I save some memory (and potentially CPU time) by storing the `micros()` into 5-8 variables of type `uint16`, discarding the upper 16-bits. This works fine because the various internal counters and events are processed more frequently than every 65.636 milliseconds. I think using a 64-bit `Duration` would be an overkill for most driver applications.

With the existing API currently, I am finding it quite difficult to create device drivers with reasonably accurate timing across multiple processors. For example:

- on AVR processors:
    -  `time.Now().Sub()` takes about 1000 microseconds, so which makes it mostly useless for writing bit banging drivers. I would guess that much of that overhead is caused by the 64-bit integers
    - `time.Sleep()` has a ~12 microsecond overhead on an AVR processor, which is better, but not sufficient for drivers requiring resolutions < 10 microseconds
- on SAMD21 processors:
    - `time.Now().Sub()` and `time.Sleep()` seem to have a minimum resolution of [213.6 microseconds](https://github.com/tinygo-org/tinygo/blob/release/src/runtime/runtime_atsamd21.go#L322)
    - I'm not able to write my device drivers on the SAMD21 using `time.Now().Sub()` and `time.Sleep()`

aykevl, 2023-05-03:
Many chips use a 32768Hz clock, so if you need better precision than that (about 30.5us) then you need something other than a timer based on the internal clock.

> Operations on `uint64` are at least 2X more expensive for additions/subtractions. For 32-bit processors without hardware support for 64-bit multiplications and divisions, `uint64` multiplications and divisions can be up to ~10X~ 16X more expensive according to my microbenchmarks.

True, but the point is to wait for a bit so it doesn't have to be particularly fast. Add/sub operations will fall well within the possible range. Multiplication most likely too. Division _might_ be slower, but I'm not entirely convinced it'll cause a noticeable slowdown.

aykevl, 2023-05-03:
Working on a precise timing package.

bxparks, 2023-05-03:
Here are my microbenchmarks on `u32` versus `u64` integer operations. (There's probably some overhead of the `volatile.LoadUint32()`, `volatile.LoadUint64()` in those numbers, but this is probably a good first-order approximation).

On some microcontrollers, `u64*u64` is 16X more expensive. I suspect that it is decomposing the `u64*u64` into 16 of the `u16*u16 => u32` multiplications, by breaking down the u64 into 4 chunks of u16.

Faster operations using u32 means less jitter in bit-banging protocols, and more CPU time available for doing other things while I'm waiting for a few microseconds. I don't need to do that very often, but it would be a nice-to-have.

I also matched my API to the Arduino API, which I think would make it easier to port libraries over from C++ to TinyGo. 

```
u32+u32: loop1: 45 loop2: 58 perOp: 1300 ns
u32*u32: loop1: 211 loop2: 272 perOp: 6100 ns
u32/u32: loop1: 405 loop2: 775 perOp: 37000 ns
u64+u64: loop1: 74 loop2: 98 perOp: 2400 ns
u64*u64: loop1: 560 loop2: 1286 perOp: 72600 ns
u64/u64: loop1: 448 loop2: 862 perOp: 41400 ns

xiao (XIAO M0/SAMD21):
u32+u32: loop1: 126 loop2: 157 perOp: 62 ns
u32*u32: loop1: 126 loop2: 157 perOp: 62 ns
u32/u32: loop1: 1343 loop2: 2612 perOp: 2538 ns
u64+u64: loop1: 199 loop2: 262 perOp: 126 ns
u64*u64: loop1: 692 loop2: 1186 perOp: 987 ns
u64/u64: loop1: 2287 loop2: 4544 perOp: 4514 ns

d1mini (ESP8266 D1 Mini)
u32+u32: loop1: 75 loop2: 100 perOp: 50 ns
u32*u32: loop1: 81 loop2: 106 perOp: 50 ns
u32/u32: loop1: 200 loop2: 343 perOp: 286 ns
u64+u64: loop1: 150 loop2: 218 perOp: 136 ns
u64*u64: loop1: 537 loop2: 962 perOp: 850 ns
u64/u64: loop1: 1118 loop2: 2168 perOp: 2100 ns

esp32-coreboard-v2 (ESP32 DevKit v1):
u32+u32: loop1: 187 loop2: 250 perOp: 31 ns
u32*u32: loop1: 175 loop2: 237 perOp: 31 ns
u32/u32: loop1: 200 loop2: 250 perOp: 25 ns
u64+u64: loop1: 325 loop2: 475 perOp: 75 ns
u64*u64: loop1: 337 loop2: 512 perOp: 87 ns
u64/u64: loop1: 1137 loop2: 2137 perOp: 500 ns
```

aykevl, 2023-05-03:
Here is the implementation: https://github.com/tinygo-org/drivers/pull/562

@bxparks I'm sidestepping all these issues by making sure the compiler can calculate the number of loops at compile time. There is no need to complicate the API if you can make a nice API while letting the compiler optimize away all expensive divisions and such.

deadprogram, 2023-10-26:
This was already released so now closing. Thanks!

closed 2023-10-26 by deadprogram