tinygo-org/drivers #564: Driver for Adafruit Seesaw & Neotrellis

trichner, 2023-05-07:
As far as I can see there is currently no support for the Adafruit Seesaw chip which powers a lot of Adafruit boards. Among them the NeoTrellis 4x4 keypad with RGB background lights.

I'd like to add partial support for the Adafruit Seesaw, namely what is used by NeoTrellis as well as NeoTrellis support.

Reference:
- [Neotrellis](https://learn.adafruit.com/adafruit-neotrellis/overview)
- [adafruit-seesaw-atsamd09-breakout.pdf](https://github.com/tinygo-org/drivers/files/11414864/adafruit-seesaw-atsamd09-breakout.pdf)

According to the contribution guidelines I'm supposed to open an issue first :)

Is there already support that I missed?

conejoninja, 2023-05-07:
Hello!

As far as I know, there's no work on the seesaw yet, so feel free to continue and update this issue once the PR is ready.

Thanks!

trichner, 2023-05-07:
Thanks for the quick reply, I'll open a PR here as soon as it's 'presentable'.

closed 2024-02-08 by trichner