tinygo-org/drivers #573: SetScroll does not work well with NO_ROTATION in ST7789

sago35, 2023-05-23:
The reproduction method is as follows.

1. Change the following examples/initdisplay/gopher-badge.go setting to NO_ROTATION
2. tinygo flash --target gopher-badge --size short examples/basic

https://github.com/tinygo-org/tinyterm/tree/b6dd97f222d168362e41097602a4ecb14018e635

sago35, 2023-05-23:
As long as ROTATION_180 is used, it works fine.

aykevl, 2023-05-23:
`SetScrollArea` and `SetScroll` do not currently respect screen rotation.

My suggested fix is to add new APIs that look like this (and similar for horizontal scroll in 90° and 270° rotated screens):

```go
// Set up a vertical scroll area, with top and bottom being the number of pixels
// from the top or bottom that are fixed (do not scroll).
// This returns ErrInvalidScroll if the display does not currently support vertical scrolling,
// for example when it is rotated 90° or 270°.
func SetVerticalScrollArea(top, bottom int16) error

// Set the vertical scroll amount in pixels.
// For example, SetVerticalScroll(10) rotates the scrollable area around by 10 pixels.
func SetVerticalScroll(pixels int16) error
```

Notes:

  * It should respect the screen rotation, so that if the screen is rotated 180° then top and bottom are swapped for example (the behavior should be the same).
  * It should respect the row/column offset.
  * For ease of use, I think `SetXxxScroll` should also include the fixed top area so that `SetXxxScroll(10)` scrolls it down by 10 pixels, regardless of screen rotation, row/column offset, etc.

This can't easily be retrofitted into `SetScroll` because there is no way to know which direction is valid. It would be possible to at least fix it for 180° rotated screens though.

I wanted to add such an API a while ago for my own needs but haven't done so yet. I'll probably look into it soon.

aykevl, 2023-06-15:
Ran into this issue myself. It looks like the rotation configuration on the st7789 is in fact not correct: it seems like all rotation values are rotated by 180° (which is why 180° would work but 0° not). And of course, SetScrollArea doesn't respect the rotation.

I plan on making a PR to fix `SetScrollArea` and `SetScroll` to respect screen rotation, and to fix the rotation to be the actual rotation of the screen (and not off-by-180°). One step in that direction is #581.

aykevl, 2023-06-19:
I have a patch ready, but it depends on a few other patches.

aykevl, 2023-06-20:
Here is a fix: https://github.com/tinygo-org/drivers/pull/584

closed 2023-09-21 by deadprogram