tinygo-org/drivers #579: Update documentation for ssd1351 driver

AWDDude, 2023-06-09:
I’m trying to use the example code for the ssd1351.  The ‘ssd1351.New()’ function wants the enPin and rwPin. I can’t find what those pins are referring to anywhere in documentation. Can someone help me out?

AWDDude, 2023-06-15:
Bump,hoping to get some information on how to use this driver.

graugans, 2023-11-09:
I guess @yannishuber can shed some light on this. When your module does not provide those PINs I assume it is save to use the `machine.NoPin` constant for those PINs

AWDDude, 2023-11-13:
> I guess @yannishuber can shed some light on this. When your module does not provide those PINs I assume it is save to use the `machine.NoPin` constant for those PINs

Thanks for the reply. I'll try that out and report back in a few days.