tinygo-org/drivers #588: ws2812: 12V-driven WS2811 LEDs flicker

diamondburned, 2023-07-19:
PR https://github.com/tinygo-org/drivers/pull/313 apparently adds support for WS2811 driven at 5V, but does 12V change this?

I'm currently writing a test program similar to the one in the example and flashing it onto an ESP32, but the LEDs just flicker:

https://github.com/tinygo-org/drivers/assets/8463786/4d8562e8-f1d1-473f-8bf5-cd2431175cae

---

<details>
<summary>Reproducing code</summary>

```go
package main

import (
	"image/color"
	"machine"
	"time"

	"tinygo.org/x/drivers/ws2812"
)

const numLEDs = 50

var colors [numLEDs]color.RGBA

func main() {
	machine.GPIO1.Configure(machine.PinConfig{Mode: machine.PinOutput})

	led := ws2812.New(machine.GPIO1)
	var state bool

	for {
		for i := range colors {
			if state {
				colors[i] = color.RGBA{255, 0, 0, 255}
			} else {
				colors[i] = color.RGBA{0, 0, 0, 255}
			}
		}
		led.WriteColors(colors[:])
		state = !state
		time.Sleep(time.Second)
	}
}
```

</details>

<details>
<summary>Build command</summary>

```
sudo tinygo flash -target=esp32-coreboard-v2 -port=/dev/ttyUSB0
```

</details>

<details>
<summary>Product information</summary>

- ESP32: https://www.amazon.com/gp/product/B0B9G74NNM
- WS2811 LEDs: https://www.amazon.com/gp/product/B01AG923EU

</details>

aykevl, 2023-07-19:
Hmm, it's really hard to say what exactly goes wrong here. This could be so many issues.

As a first step, can you try disabling interrupts?

```go
import "runtime/interrupt"

...

state := interrupt.Disable()
led.WriteColors(colors[:])
interrupt.Restore(state)
```

Then, there could be many reasons for this particular issue, like:

1. Insufficient power (60mA * 50 LEDs = 3A of power needed!). Are you sure you provide the LEDs with enough power, using a thick wire? (No breadboard wires etc).
2. Have you connected the ground of the power source and of the ESP32? It's hard to see from the video, but I don't think I see a ground wire.
3. Is it a reasonably short data wire? Standard breadboard wires should be fine, more than a meter of wire can be problematic.

diamondburned, 2023-07-19:
> Have you connected the ground of the power source and of the ESP32? It's hard to see from the video, but I don't think I see a ground wire.

This ended up being the problem... Sorry for wasting your time...

closed 2023-07-19 by diamondburned

aykevl, 2023-07-20:
Good to hear, happy to help :)