tinygo-org/drivers #630: [HUB75] ABC/ABCDE panels support

funkycode, 2023-12-20:
https://github.com/tinygo-org/drivers/blob/fd21e6ac9b4a5c3bd906ed8a82783a972be6d4ca/hub75/hub75.go#L60

The implementation supports only ABCD panels, while there are some panels that have only ABC and moreover from my understanding there are also panels with ABCDE pins.

I have ABC only panel on it's way from AliExpress that I want to test, if needed to adjust this implementation and make a PR once it is working.

Should we have multiple implementations or should we add something like ENUM to define which type it is?

[Here](https://github.com/hzeller/rpi-rgb-led-matrix/issues/120#issuecomment-209547310) is thread for reference on differences and implementation o other project that I am going to use as help to adjust tinygo implementation for ABC only support (can't make ABCDE one as I lack the led matrix of that type)

conejoninja, 2023-12-20:
Depending on how much difference there are between versions.

Personally, I would add a new constructor... like ```func NewABC(b drivers.SPI, latPin, oePin, aPin, bPin, cPin machine.Pin) Device {```

and use some sort of variable to keep track which panel type is and how to handle the pins. That's assuming there is not much difference and we could use the hub75 package for both. If there's a major difference on how to handle the code, then I'll vote for _hub75_ (as is) and a new _hub75abc_ driver

funkycode, 2023-12-25:
Well, at the end one I got utilizes all 4 channels. I have another matrix that is on the way, in case it will use only ABC I will try to implement it, in case not - unfortunately i would not be able to do so as I would need HW to test implementation :(