tinygo-org/drivers #636: net, wioterminal: connect to access point claims success even if it has failed

deadprogram, 2024-01-10:
Tried to use the `wioterminal` with the latest `dev` branch, and it claimed to connect to a WiFi access point that was not actually turned on at the time. :ghost: 

```
$ tinygo flash -size short -target wioterminal -ldflags="-X main.ssid=TELLO-C48E59" -monitor ./examples/takeoff/
   code    data     bss |   flash     ram
 105924    2508   10072 |  108432   12580
Connected to /dev/ttyACM0. Press Ctrl-C to exit.
RTL8720 firmware version : 2.1.2
MAC address              : 2c:f7:f1:1b:48:35

Connecting to Wifi SSID 'TELLO-C48E59'...CONNECTED
```

scottfeldman, 2024-01-10:
Minimum wifi passphrase is 8 chars...I think that's the root of the problem, but need to look at the logic that checks the connection...

scottfeldman, 2024-01-11:
Fixed in https://github.com/tinygo-org/drivers/pull/637

deadprogram, 2024-02-28:
Closing as completed in the most recent release.

closed 2024-02-28 by deadprogram