tinygo-org/drivers #658: Unable to Build examples/net/http-get on Windows

sago35, 2024-02-26:
I couldn't build examples/net/http-get on Windows.

```
$ tinygo version
tinygo version 0.31.0 windows/amd64 (using go version go1.22.0 and LLVM version 17.0.1)

$ tinygo build -size short -o ./build/test.hex -target=pyportal -stack-size 8kb ./examples/net/http-get/
panic: interp: offset out of range

goroutine 285 [running]:
github.com/tinygo-org/tinygo/interp.pointerValue.addOffset(...)
        C:/tinygo/tinygo/interp/memory.go:523
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc01789e3c0, 0xc0172e25f0, {0xc01f128510, 0x2, 0x0?}, 0xc01f116f60, {0xc01f10ff30, 0x10})
        C:/tinygo/tinygo/interp/interpreter.go:661 +0x8eee
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc01789e3c0, 0xc0172e25a0, {0xc01f128450, 0x3, 0x0?}, 0xc01f116e40, {0xc01f10ff20, 0xc})
        C:/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc01789e3c0, 0xc01ef0d770, {0xc01f0fea50, 0x1, 0x0?}, 0xc01f0fcd20, {0xc01f10fed8, 0x8})
        C:/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc01789e3c0, 0xc01ef0d4a0, {0x0, 0x0, 0xc0173d01c0?}, 0x0, {0x7ff79a5c0515, 0x4})
        C:/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.Run({0x7ff79572003b?}, 0x1a1e6926d90?, 0x0?)
        C:/tinygo/tinygo/interp/interp.go:117 +0x68c
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x1a192657c40?}, 0xc0002a6000, 0xc000294210)
        C:/tinygo/tinygo/builder/build.go:1034 +0x3c
github.com/tinygo-org/tinygo/builder.Build.func5(0xc016c63b00?)
        C:/tinygo/tinygo/builder/build.go:577 +0x64d
github.com/tinygo-org/tinygo/builder.runJob(0xc016c63b60, 0xc0164c9c80)
        C:/tinygo/tinygo/builder/jobs.go:222 +0x4d
created by github.com/tinygo-org/tinygo/builder.runJobs in goroutine 1
        C:/tinygo/tinygo/builder/jobs.go:123 +0x5bd
```

sago35, 2024-02-26:
I encountered the same result even when using the official release. Next, I'll try using Go version 1.21.

```
$ /c/tinygo/tinygo0.31.0.windows-amd64/tinygo/bin/tinygo build -size short -o ./build/test.hex -target=pyportal -stack-size 8kb ./examples/net/http-get/
panic: interp: offset out of range

goroutine 43 [running]:
github.com/tinygo-org/tinygo/interp.pointerValue.addOffset(...)
        D:/a/tinygo/tinygo/interp/memory.go:523
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc014b06000, 0xc02038abe0, {0xc01cc8cad0, 0x2, 0x0?}, 0xc0067bd560, {0xc011459930, 0x10})
        D:/a/tinygo/tinygo/interp/interpreter.go:661 +0x8eee
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc014b06000, 0xc02038aaf0, {0xc01cc8ca10, 0x3, 0x0?}, 0xc0067bd3e0, {0xc011459920, 0xc})
        D:/a/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc014b06000, 0xc01af00af0, {0xc01b701810, 0x1, 0x0?}, 0xc00682e5d0, {0xc0114598d8, 0x8})
        D:/a/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc014b06000, 0xc01ce75c70, {0x0, 0x0, 0x1401ec2c9?}, 0x0, {0x1451fc4b5, 0x4})
        D:/a/tinygo/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.Run({0x1403bfcbb?}, 0x5e7f4f80?, 0x0?)
        D:/a/tinygo/tinygo/interp/interp.go:117 +0x68c
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x4bbc1300?}, 0xc0002a6000, 0xc000294210)
        D:/a/tinygo/tinygo/builder/build.go:1034 +0x3c
github.com/tinygo-org/tinygo/builder.Build.func5(0xc01937b140?)
        D:/a/tinygo/tinygo/builder/build.go:577 +0x64d
github.com/tinygo-org/tinygo/builder.runJob(0xc01937ade0, 0xc01b2ed560)
        D:/a/tinygo/tinygo/builder/jobs.go:222 +0x4d
created by github.com/tinygo-org/tinygo/builder.runJobs in goroutine 1
        D:/a/tinygo/tinygo/builder/jobs.go:123 +0x5bd
```

sago35, 2024-02-26:
I was able to build it using Go 1.21.7.

```
$ /c/tinygo/tinygo0.31.0.windows-amd64/tinygo/bin/tinygo version
tinygo version 0.31.0 windows/amd64 (using go version go1.21.7 and LLVM version 17.0.1)

sago3@B450GT3 MINGW64 ~/dev/src/tinygo.org/x/drivers (dev)
$ /c/tinygo/tinygo0.31.0.windows-amd64/tinygo/bin/tinygo build -size short -o ./build/test.hex -target=pyportal -stack-size 8kb ./examples/net/http-get/
   code    data     bss |   flash     ram
 230500    2920    7040 |  233420    9960
```

scottfeldman, 2024-02-27:
I get the same results on Linux.  Work with go1.21.5, but fails with go1.22.0.

```
sfeldma@nuc:~/work/tinygo(dev)$ tinygo version
tinygo version 0.31.0 linux/amd64 (using go version go1.22.0 and LLVM version 17.0.6)

sfeldma@nuc:~/work/tinygo(dev)$ cd ../drivers
sfeldma@nuc:~/work/drivers(dev)$ tinygo build -size short -target=pyportal -stack-size 8kb ./examples/net/http-get/
panic: interp: offset out of range

goroutine 260 [running]:
github.com/tinygo-org/tinygo/interp.pointerValue.addOffset(...)
        /home/sfeldma/work/tinygo/interp/memory.go:523
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc008bde000, 0xc011890cd0, {0xc01f2e5190, 0x2, 0x0?}, 0xc0089f2030, {0xc02bad9570, 0x10})
        /home/sfeldma/work/tinygo/interp/interpreter.go:661 +0x8eee
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc008bde000, 0xc00e823810, {0xc01f2e5090, 0x3, 0x0?}, 0xc0089adce0, {0xc02bad9560, 0xc})
        /home/sfeldma/work/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc008bde000, 0xc019577db0, {0xc01f2b4b50, 0x1, 0x0?}, 0xc00817d1a0, {0xc02bad9518, 0x8})
        /home/sfeldma/work/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc008bde000, 0xc016704af0, {0x0, 0x0, 0xd0?}, 0x0, {0x8d40cf, 0x4})
        /home/sfeldma/work/tinygo/interp/interpreter.go:529 +0x7c85
github.com/tinygo-org/tinygo/interp.Run({0x7bbdbb?}, 0x7fbf1cee6b00?, 0x0?)
        /home/sfeldma/work/tinygo/interp/interp.go:117 +0x68c
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x7fbf402c16d0?}, 0xc00025ae80, 0xc000275b00)
        /home/sfeldma/work/tinygo/builder/build.go:1034 +0x3c
github.com/tinygo-org/tinygo/builder.Build.func5(0xc013b388a0?)
        /home/sfeldma/work/tinygo/builder/build.go:577 +0x62d
github.com/tinygo-org/tinygo/builder.runJob(0xc0118523c0, 0xc00dee3800)
        /home/sfeldma/work/tinygo/builder/jobs.go:222 +0x4d
created by github.com/tinygo-org/tinygo/builder.runJobs in goroutine 1
        /home/sfeldma/work/tinygo/builder/jobs.go:123 +0x5bd
```

scottfeldman, 2024-02-27:
I ran smoketest.sh and got some weird results...some of the examples/net examples pass and others fail.

```
ntpclient          pass
http-get           FAIL
tcpclient          pass
websocket/dial     FAIL
socket             pass
webstatic          FAIL
tlsclient          FAIL
mqttclient/natiu   pass
webclient          pass
webserver          FAIL
```

Not sure what that tells us...have to think about it.

deadprogram, 2024-02-29:
This should now all be resolved with `0.31.1`

deadprogram, 2024-05-06:
Now closing, thanks for reporting @sago35

closed 2024-05-06 by deadprogram