tinygo-org/drivers #660: Driver contribution: DFRobot's DFPlayer Mini MP3 player driver

0b1-k, 2024-02-28:
I'm working on an embedded project requiring MP3 / .WAV file playback and I wrote a driver for the DFPlayer Mini MP3 player chip that I'm using for that purpose. I thought that this driver might be useful to other TinyGo users. 

You can find the module here: https://github.com/0b1-k/DFPlayerMini

conejoninja, 2024-03-01:
Yes! I've just started my own driver because I'm using the MP3Player for a thing, but since you already have done it, you saved me a lot of time and work!! thank you!

Could you add it to the repo? That would be great and very useful. If not, I could add it based on your work.

0b1-k, 2024-03-01:
Hi Daniel,

I will add it to the repo :)

0b1-k, 2024-03-01:
Daniel,

I've created pull request #664 to include this driver to the repo.
Please let me know if you have any questions or issues.