tinygo-org/drivers #685: compiler error while importing tinygo.org/x/drivers/ssd1306: expected 'package', found 'EOF'

BryanSouza91, 2024-06-11:
I get this error when trying to flash:
```
$ tinygo build -target=nodemcu -port=/dev/ttyUSB0 ../ssd1306      
../../../go/pkg/mod/tinygo.org/x/drivers@v0.27.0/ssd1306/registers.go:1:1: expected 'package', found 'EOF'
```

I have tried many solutions found when searching for this error. I have run into this error previously with none tinygo projects and was able to fix the issue however I am unable to do so with this issue.

go.mod

```
module ssd1306
go 1.22.3
require tinygo.org/x/drivers v0.27.0
```

go.sum
```
tinygo.org/x/drivers v0.27.0 h1:TEGk1lQvEhXxfvpEhUu+pwmCnhtldPI+hpHlO9VYixI=
tinygo.org/x/drivers v0.27.0/go.mod h1:q/mU8G/wz821p8xXqbkBACOlmZFDHXd//DnYnCW+dDQ=
```

```
$ tinygo info -target=nodemcu
LLVM triple:       xtensa
GOOS:              linux
GOARCH:            arm
build tags:        xtensa baremetal linux arm esp8266 esp nodemcu tinygo purego math_big_pure_go gc.conservative scheduler.tasks serial.uart
garbage collector: conservative
scheduler:         tasks
cached GOROOT:     /home/bryansouza/.cache/tinygo/goroot-ef106e068c9f67ad57092cc9befb16284ba02816acb59341c17f2fda4a265afd
```

aykevl, 2024-06-13:
It looks like there's something wrong with your system that caused the file to be empty. Because [it really isn't](https://github.com/tinygo-org/drivers/blob/v0.27.0/ssd1306/registers.go).
Can you try `go clean -modcache` and see if that helps?

BryanSouza91, 2024-06-13:
I ran `go clean -modcache` with no change. I also reinitialized my module multiple times cleaning the modcache each time. You are correct the file is not empty. I even deleted the registers.go file and remade it with no change.

BryanSouza91, 2024-06-13:
I just checked it again today and it seems to be working correctly now. However I am getting errors from the machine package.I have to figure out how to properly install the Tinygo packages on my system...

closed 2024-06-13 by BryanSouza91