tinygo-org/drivers #691: [HUB75]Leds do not light up as expected

ehime-iyokan, 2024-07-02:
I wrote program using hub75 package. but, it didn't light up. So I have tried to fix hub75.
As a result. It light up as expected. And Environmental information is here. -> https://github.com/ehime-iyokan/tinygo_hub75_sample

[image]

First, I added a `time.sleep(1 * time.Microsecond)` after latch-High.
By doing so, the light came on. However, I don't know why, so if you know, please let me know.
Moreover, the red lights are on in different places.

[image]

Second, I fixed buffering of red data.
I guess it has to do left-shift first. 
`d.buffer[c][offsetR] = d.buffer[c][offsetR] &^ 1 << bitSelect`
-> `d.buffer[c][offsetR] &^= 1 << bitSelect`

[image]

ehime-iyokan, 2024-07-02:
`I added a time.sleep(1 * time.Microsecond) after latch-High.
By doing so, the light came on. However, I don't know why`
If I figured out why it worked correctly, I'll create a pull request.

sago35, 2024-07-13:
@ehime-iyokan 
The GPIO of the RP2040 might be too fast. Without adding time.Sleep(), the pulse width is 110ns, which might be too short. With time.Sleep(), it is around 5us.

I think the fix you are considering is necessary.

## with `time.Sleep(1 * time.Microsecond)` : 5.046us
[image]

## without `time.Sleep(1 * time.Microsecond)` : 110ns
[image]

sago35, 2024-07-13:
Including other parts as well, it is better to use delay.Sleep for microsecond sleeps. I think it would be good if you could create a Pull Request with that change.

https://github.com/tinygo-org/drivers/blob/v0.28.0/delay/sleep.go

## with `delay.Sleep(1 * time.Microsecond)` : 1.16us

[image]

ehime-iyokan, 2024-11-04:
I created pullrequest. => (https://github.com/tinygo-org/drivers/pull/722)

```
Second, I fixed buffering of red data.
I guess it has to do left-shift first.
d.buffer[c][offsetR] = d.buffer[c][offsetR] &^ 1 << bitSelect
-> d.buffer[c][offsetR] &^= 1 << bitSelect
```
(https://github.com/tinygo-org/drivers/issues/691#issue-2386396643)

ehime-iyokan, 2024-11-04:
Thank you for providing the information.  @sago35 
For now, I've decided to use a microcontroller board other than the RP2040. Next time I use the RP2040, I’ll consider creating a pull request.

closed 2025-08-11 by deadprogram