tinygo-org/drivers #694: User servo with Pico

eleimt, 2024-07-19:
Hello
Tell me please, how use servo.go with Pico? I have this code:
```go
package main

import ...

func main() {
	srv, err := servo.New(machine.PWM0, machine.GP16)
	if err != nil {
		println(err.Error())
	}

	for i := 0; i <= 180; i++ {
		println("I:", i)
		srv.SetAngle(i)
		time.Sleep(time.Millisecond * 10)
	}
}
```

But servo works in jerks..

[image]

deadprogram, 2024-07-20:
@eleimt please see Servo example in repo:

https://github.com/tinygo-org/drivers/blob/release/examples/servo/servo.go

eleimt, 2024-07-20:
@deadprogram The situation has not changed

eleimt, 2024-07-20:
@deadprogram Hi, thanks, I find problem. Problem was not in code.

closed 2024-07-20 by eleimt