tinygo-org/drivers #696: dht package is broken

Gnyblast, 2024-07-23:
Since release [v0.20.0](https://pkg.go.dev/tinygo.org/x/drivers@v0.20.0/dht) when `device.go` introduced to the package, the methods are gone for dht package like: `dht.New()`. dht package now only exposes `device.go` constants and types which just 3 items, as you can see in the image below.

[image]

[v0.19.0](https://pkg.go.dev/tinygo.org/x/drivers@v0.19.0/dht) seem to not having this issue though.

deadprogram, 2024-07-23:
Hello @Gnyblast I think the issue must be with the auto-complete in your editor, since the `dht` package itself builds correctly in the CI.

See https://github.com/tinygo-org/drivers/blob/release/smoketest.sh#L101 and https://github.com/tinygo-org/drivers/actions/runs/9563791661/job/26363092847#step:8:378

Perhaps you need to open an issue in https://github.com/tinygo-org/vscode-tinygo?

Gnyblast, 2024-07-23:
Yeah since addition of go tags to top `//go:build tinygo`, gopls stopped working.

Gnyblast, 2024-07-23:
Yeah problem was vscode-tinygo configs from the looks of it. Thanks.

closed 2024-07-23 by Gnyblast

Gnyblast, 2024-07-24:
Sorry to coming back to this again but let me explain. I'm writing a program for especially for Raspberry pi **(not pico)** models. And I'm not using tinygo because it's obviosuly running an OS not a microcontroller. I just wanted to take advantage of your `drivers` module to create a factory for inpur devices, communicate and fetch data from them. But obviously since `v0.20.0` the addition of `//go:build tinygo` prevents default `gopls` on vscode to resolve the code correctly also the go compiler as well. This go tag is added because the code is not compatible with main go binary for compilation and being used as library or just to make things more integrated on tinygo side?