tinygo-org/drivers #699: go.mod weirdness or my stupidity?

gherlein, 2024-08-13:
If I have a go.mod with:  

module tinygo.org/x/drivers

I get this error:

```bash
gherlein@io:~/src/tinygo/m$ go mod tidy
go: finding module for package tinygo.org/x/drivers/mcp2515
go: tinygo.org/x/drivers imports
	tinygo.org/x/drivers/mcp2515: cannot find module providing package tinygo.org/x/drivers/mcp2515: unrecognized import path "tinygo.org/x/drivers/mcp2515": reading https://tinygo.org/x/drivers/mcp2515?go-get=1: 404 Not Found
```
If I change the go.mod to

module tinygo.org/drivers

it works.  

I'm too new to go.mod foo to really grok this but it does not seem right.

deadprogram, 2024-08-14:
See https://github.com/tinygo-org/tinygo-site/pull/419 for PR that adds the missing files.

gherlein, 2024-08-14:
so fast!  Great.  How do I learn more about the module index stuff?  It seems that tinygo.org/x/drivers/mcp2515 is somehow pointing to https://tinygo.org/x/drivers/mcp2515?go-get=1 - which you fixed the redirect.  But how do I get smarter on where the index lives that does that mapping?  If you know I'd love a pointer so I get smarter.

thanks again for the very fast fix.

deadprogram, 2024-08-14:
See https://pkg.go.dev/cmd/go#hdr-Remote_import_paths for more info

gherlein, 2024-08-14:
> See [tinygo-org/tinygo-site#419](https://github.com/tinygo-org/tinygo-site/pull/419) for PR that adds the missing files.

If I get a wild hair and want to propose some documentation updates, are you guys open to PRs for the web site?

deadprogram, 2024-08-14:
Absolutely yes!

gherlein, 2025-03-26:
@deadprogram  it seems broken again:

io:~/src/tinygo/pico/sx127x> go mod tidy
go: errors parsing go.mod:
go.mod:5: unrecognized import path "tinygo.org/x/drivers/sx127x": reading https://tinygo.org/x/drivers/sx127x?go-get=1: 404 Not Found

changing to drop the x:
io:~/src/tinygo/pico/sx127x> go mod tidy
go: errors parsing go.mod:
go.mod:5: unrecognized import path "tinygo.org/drivers/sx127x": reading https://tinygo.org/drivers/sx127x?go-get=1: 404 Not Found

Is this me?