tinygo-org/drivers #718: Add support for LSM303DLHC e-Compass

BryanSouza91, 2024-10-29:
I have an Adafruit breakout board for the LSM303DLHC e-Compass. Is anyone working on this? I didn't find an open issue. 

I also see that the LSM303AGR is supported and that will be the basis of adding the LSM303DLHC.

deadprogram, 2024-10-29:
Sounds good, thank you!

BryanSouza91, 2024-10-29:
I ran the tests but received an error on the final build test since my version of Go is 1.23 and the test wants versions 1.18 through 1.22, is this an issue? I also have not actually tested the code with the device, should I try that prior to the pull request?(probably yes)

BryanSouza91, 2024-11-02:
I updated tinygo to the latest version and the previous error is fixed however there seems to be another error with the smoketest. I have not been able to build a small test program with the new lsm303dlhc code because the code is not part of the official repo. I will submit the pull request and go from there. 

This can be closed if this issue is complete at this point.

BryanSouza91, 2025-08-09:
Refer to [PR #720](https://github.com/tinygo-org/drivers/pull/720)