tinygo-org/drivers #730: ST7789 parallel?

foxt, 2024-12-18:
Hi, I was looking to see if I could try TinyGo on a Pimoroni Tufty2040, I see there's already a [board configuration](https://tinygo.org/docs/reference/microcontrollers/tufty2040/) for it, but from what I can see the existing ST7789 only supports SPI mode whereas on the Tufty the LCD is connected in parallel mode? Is there any plans to support ST7789 in parallel mode? Thanks in advance

aykevl, 2024-12-19:
Is the LCD only available in parallel mode, or can you also control it using SPI?
The RP2040 is often fast enough to control the display over SPI. So if that works, you can try that first.

I'm not aware of any ST7789 parallel support in TinyGo, but you can take a look at the ILI9341 for a driver that does support this.

foxt, 2024-12-19:
@deadprogram yeah thanks! that seems to be what i was looking for, however i couldn't seem to get it working. no errors on serial console, and the backlight turned on but the lcd was either just blank or occasionally garbled data

soypat, 2025-07-24:
I missed this issue! I implemented parallel for the RP2 architecture a while ago helping David De Leon get it up and working for the tufty! I'm not sure it was throughly tested- help testing is much appreciated, I am available to perform fixes needed to get it working

https://github.com/tinygo-org/pio/blob/main/rp2-pio/piolib/parallel8.go

soypat, 2025-07-31:
If anyone has a tufty+rp2040+logic analyzer I'd ask you to capture the parallel transaction over the 8 wire bus and send it to me so I can study whatever bug is present in the PIO implementation of the parallel8 protocol

soypat, 2025-07-31:
@conejoninja Hey! so what i need is the following setup:
- Tufty connected to RP2040 pins following schema in example: https://github.com/tinygo-org/pio/blob/main/rp2-pio/examples/tufty/tufty.go
   - Keep in mind `db0Pin` points to the start of the consecutive block of pins which constitute the 8 parallel pins. So if db0 is GPIO14, then GPIO15 will be db1, GPIO16 will be db2... all the way up to db7 which is GPIO21. If you are using the tufty2040 you need not change anything in the example
- Hook up your digital analyzer to the parallel 8 pins, whichever db0 though db7 end up being and run the example. This is PARALLEL8 capture. We are not done yet!
- hook up your digital analyzer to csPin, dcPin, wrPin, rdPin and blPin and hook up remaining free logic analyzer inputs to any of the db0-7 pins you have. Run the example. this is PARALLELCTL capure.
- Finally send me the captures noting which pin is which. If you are using Logic 2 by Saleae Pro you can name the pins so it solves the problem for me

As for the PIO-free driver, that is absolutely possible- that said, I doubt it has much practical use due to the strain on the CPU it would have. Not only does the RP2040 provide the PIO to do clock-cycle pulses, but it also provides DMA, practically freeing up the CPU during operation. Maybe when TinyGo adds multicore for more CPUs. Keep in mind these are screens, not any old driver. Bitbang protocols would cause very noticeable slowdown. For comparison, the PIO implementation of the wifi driver sped up the initialization from 14 seconds down to 200ms. That's a x70 speedup. For comparison, if your screen updates 24 times per second using PIO (a decently good refresh rate for a screen), you may get slowed down to one refresh every 3 seconds! This is not considering the higher overhead of a screen refresh compared to wifi driver exchanges, which are on the order of a few hundred bytes.

diamondburned, 2025-07-31:
> One recommendation: if it's not working with Tinygo, I suggest you test it with Pimoroni's examples (C++ or MicroPython). To make sure the HW is ok.

It's definitely working! I got the Tufty2040 from a friend who had her own configured Micropython firmware and it was displaying fine.

I unfortunately do not have a logic analyzer since I'm really new to embedded programming stuff still, but I'd be happy to test anything new!