tinygo-org/drivers #733: Digital Motion Processor device driver extension  for mpu6050

brucebotes, 2025-01-08:
This is my first ever contribution to the open source community. So please  bear with me and help with the contribution if it does add benefit/ seem fit.

It is a tinygo transaltion of the Jeff Rowberg C/C++ arduinno library for the DMP firmware. I am not sure if this is acceptable and/or legit. If there are no issues,  the team is welcome to add the extension to the MPU6050 library. 

If someone is prepared to guide me through the process I am prepared to give it a shot. 

Of note is the video and  python animation as provided by Maker's Wharf. Please see the referenced youtube link. 
How would we incorporate that within the existing setup?

Please find the repo here: https://github.com/brucebotes/tinygo_mpu6050_dmp

Ps. I tried this Slack link (https://gophers.slack.com/messages/CDJD3SUP6.)to raise this quesion, but that does not seem to work. So here goes...

aykevl, 2025-09-30:
Ideally we'd just put it all in the same driver folder - the mpu6050 folder itself. I think that will make the package easier to use.

brucebotes, 2025-10-02:
Hi Ayke,
Done. I hope and cross my fingers that all is well with the PR. This was my
first PR contribution to open source.
Cheer
Bruce