tinygo-org/drivers #739: Drivers for Qwiic & StemmaQT boards?

jphastings, 2025-02-14:
I have a number of Qwiic & StemmaQT boards that I'd be interested in getting working with Tinygo. I've not built any drivers before, but I may try and build them if I can't find them!

I'll update this issue with what I find as I go, but **I'd appreciate pointers to drivers that already exist**.

(If you'd _also_ like a driver for one of these boards, please just 👍 on this issue, rather than post a comment 😊)

---

I'm testing with a [QT Py RP2040](https://tinygo.org/docs/reference/microcontrollers/qtpy-rp2040/).

| Company   | Board                                                                                                      | Purpose                      | Existing driver                                                                                  | Testing notes               |
|-----------|------------------------------------------------------------------------------------------------------------|------------------------------|--------------------------------------------------------------------------------------------------|-----------------------------|
| Elechouse | [NFC module v3](https://www.elechouse.com/elechouse/images/product/PN532_module_V3/PN532_%20Manual_V3.pdf) | NFC reader                   | [3rd party driver](https://github.com/graugans/tinygo-examples/tree/main/drivers/pn532)          | I've not had it working yet |
| Adafruit  | [STEMMA MiniGPS](https://www.adafruit.com/product/4415)                                                    | Geographic location          | [Possible official driver](https://github.com/tinygo-org/drivers/main/examples/gps/i2c/main.go)? | I've not had it working yet |
| Adafruit  | [NeoPixel driver](https://www.adafruit.com/product/5766)                                                   | RGB LED strip driver         | [Possible official driver](https://github.com/tinygo-org/drivers/tree/release/ws2812)?           | I've not had it working yet |
| Adafruit  | [ST25DV16](https://thepihut.com/products/adafruit-st25dv16k-i2c-rfid-eeprom-breakout-stemma-qt-qwiic)      | NFC tag emulator (ish)       | no driver?                                                                                       |                             |
| Adafruit  | [LTR-329](https://www.adafruit.com/product/5591)                                                           | Light sensor                 | no driver?                                                                                       |                             |
| Adafruit  | [Rotary Encoder](https://www.adafruit.com/product/5880)                                                    | Rotary encoder & LED         | no driver?                                                                                       |                             |
| Adafruit  | [Infineon Trust M](https://www.adafruit.com/product/4351)                                                  | Crypto functions & key store | no driver?                                                                                       |                             |
| Sparkfun  | [Qwiic Button](https://www.sparkfun.com/sparkfun-qwiic-button-green-led.html)                              | Button & LED                 | no driver?                                                                                       |                             |
| Sparkfun  | [Qwiic Alphanumeric Display](https://www.sparkfun.com/sparkfun-qwiic-alphanumeric-display-kit.html)        | 4 letter display             | no driver?                                                                                       |                             |

(Places I've looked: [Official drivers](https://github.com/tinygo-org/drivers), [Periph drivers](https://github.com/periph/devices) — code & issues/PRs, Github & searching the internet.)

conejoninja, 2025-02-14:
Hello, I like this, nice initiative, I might be able to help if I get some of the boards.

Qwiic & StemmaQT is just the connector for I2C, you need to look for the specific chip on those boards.

- Rotary encoder is using the [adafruit's seesaw firmware](https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout) -Adafruit seesaw is a near-universal converter framework- , there's already a driver for it ( tinygo.org/x/drivers/seesaw ), some modification might be needed to work with the rotary encoder

- The Qwiic button uses a similar library (in a sense), here's the C/Arduino library you can use as a base : https://github.com/sparkfun/SparkFun_Qwiic_Button_Arduino_Library

- Qwiic alphanumeric display uses a vk16k33 led driver

- neopixel is ws2812 driver

jphastings, 2025-02-15:
Thanks @conejoninja! I've added your information into my table, and created a column for the I2C chip driving each of these boards (great pointer) — hopefully I'll get some time over the next week or two to try out the rotary encoder & the qwiic button (which seems simple enough it should be a great starter driver project for me!)

jphastings, 2025-06-30:
I ended up getting a job (!), so didn't have time time to continue this earlier this year, but I recently discovered that the TinyGo GPS driver works with e PA1010D _if_ you use the correct i2c address. I've [opened a PR](https://github.com/tinygo-org/drivers/pull/770) to provide easy use of it. I'll keep going with the other devices when I find a spare moment!

To be honest, I'm currently very motivated to get the MMC5983MA magnetometer working, so that's likely to be the next one!