tinygo-org/drivers #743: comboat: duplicate methods "execute" and "send"

deadprogram, 2025-02-22:
The "execute" and "send" methods for this driver have the exact same implementation

https://github.com/tinygo-org/drivers/blob/06dd60fba2b24144700494ebd6d298c5bc57de74/comboat/comboat.go#L142

deadprogram, 2025-02-22:
cc @scottfeldman 😄

scottfeldman, 2025-02-22:
They are very close, but not the same.  execute() waits for "OK", send() waits for ">".  I thought about combining them into one func, but that would cause problems if a sending process see's the OK from another process that's executing a cmd, like creating another socket. 

Maybe there is a way to DRY the duplicated parts of send() and execute()?

scottfeldman, 2025-02-22:
> I thought about combining them into one func, but that would cause problems if a sending process see's the OK from another process that's executing a cmd, like creating another socket.

[Caffeine kicking in]...scratch that thought...the two competing processes are mutual ex due to d.Lock().

But I still like being explicit about expected result after an execute() cmd vs. a send() cmd.  Maybe the code just needs a // comment covering this?