tinygo-org/drivers #754: WS2812 LED stopped changing colour on Pico 2040 with 200MHz clock

leon-matthews, 2025-05-03:
My [blink program](https://codeberg.org/leon_matthews/tinygo-playground/src/branch/main/blink/rp2040-zero-ws2818/main.go) for built-in WS2812 LED on Waveshare RP2040 Zero stopped working after upgrading TinyGo from 0.36.0 to 0.37.0.

I see from the changelog that the CPU frequency for the RP2040 has been boosted from 125 to 200MHz (nice!), but the WS2812 driver doesn't seem to support such lofty frequencies. The switch statement in [ws2812/ws2812_cortexm.go](https://github.com/tinygo-org/drivers/blob/release/ws2812/ws2812_cortexm.go) has a case for 125MHz but tops out at 168MHz - there is no case for the new 200MHz CPU frequency.

My next step was to read through the docs in the `machine` package to see if I could set the CPU back to 125MHz, but could not see any way to do that.

leon-matthews, 2025-05-03:
I just downgraded TinyGo to v0.36 and I can confirm that my WS2812 example started working again - and that its CPU frequency was downgraded to 125MHz.

leon-matthews, 2025-05-04:
The RP2040's PIO hardware is mode for this sort of bit-bashing and wouldn't you know it,  the fantastic team behind TinyGo has already made a PIO driver for WS2812 LEDs! Since yesterday I have ported my [WS2812 blink](https://codeberg.org/leon_matthews/tinygo-playground/src/branch/main/rp2040-zero/blink-rainbow/main.go) program to use this PIO driver, and played around with Go's new iterators to add a rainbow effect too.

The WS2812 driver in this repo still needs a few more `nop` operations in order to work at 200MHz, but the PIO driver works right now - and is a better choice for Pico-based devices anyway.

leon-matthews, 2025-05-06:
I'm going to have a go at writing a patch to see if I can get this working. The PIO driver is clearly the best way to drive these things under a Pico, but it would be nice to have the generic driver working too. I'm a TinyGo newbie, but I'm also willing to listen and learn!

leon-matthews, 2025-05-11:
Famous last words! I found some time over the weekend to do some testing. The local tests all passed, but I got stuck writing a demo program using the new changes. I had trouble coercing TinyGo to use my local copy of the drivers module rather than the remote one. My inexperience with Go is showing - I've only picked it up this year after starting a new job.

I read through the contributors document and done some research. As far as I can see I need to add a `replace` directive into the example program's `go.mod` file. I'll give that a try as soon as I'm able then write a pull request.

jamestait, 2025-05-13:
You may want to look into [Go workspaces](https://go.dev/doc/tutorial/workspaces) for this. Unfortunately I've been sufficiently busy with other stuff to have not worked with Go for a while, otherwise I'd happily provide directions here, but I think it works similarly enough to the `replace` directive in `go.mod`, but without needing to modify `go.mod`.

deadprogram, 2025-05-20:
Labeled to close this issue after next release.

PR #759 (2025-05-20): ws2812: add 200MHz support for the Cortex-M0/rp2040
My blink program for built-in WS2812 LED on Waveshare RP2040 Zero stopped working after upgrading TinyGo from 0.36.0 to 0.37.0, after we bumped the CPU clock from 125MHz to 200MHz.

Closes #754